Edzelf/Esp-radio #178: Please add AAC & HE-AAC support

K1WIZ, 2019-10-04:
This looks like a great project and hats off to the developers.   As I understand it, the VS1053 does support decoding AAC and AAC+ (HE-AAC) streams.   Can you include support for those?  Many stations I am finding are moving to AAC+ because of the quality at much lower bandwidth.

Edzelf, 2019-10-06:
Asw far as I know, the AAC format will work on the radio.  Please send the URL of the station that does not work.

kost-1, 2020-08-06:
> 
> 
> As far as I know, the AAC format will work on the radio. Please send the URL of the station that does not work.

https://fed.fmplay.ru:8000/relax-32.aac?uid=198683153

Edzelf, 2020-08-07:
It works on ESP32 version.  It works on the ESP8266 version.  Just remove the "https://" from the station name.

Edzelf, 2020-08-07:
Maybe you have a fake VS1053 chip.  Try to read the printing on the IC.

kost-1, 2020-08-08:
Sorry, it vs1003 chip. My mistake.

Derek-K, 2021-07-02:
Tested. "fed.fmplay.ru:8000/relax-32.aac?uid=198683153" works on ESP-Radio as well. :)

m040601, 2021-07-03:
> As far as I know, the AAC format will work on the radio

@Edzelf

Could you update the README front page ? Only one sentence more or just type a few characters to make this crystal clear for end users finding this page for the first time.

Right now when one lands on the front page on bumps into this:

```
NOTES:

    If you are using V2.4.0 of the core library: set IwIP Variant to "V1.4 Prebuilt" in the Tools of the IDE.
    The radio will NOT play AACP streams.

Features:

    Can connect to thousands of Internet radio stations that broadcast MP3 or Ogg audio streams.

```

It then keeps talking a lot about "mp3" and "ogg" without any **explicit reference to aacp and it's variants**

I've been checking this project on and off for some time in the last months.
I've search the documentation extensevely.

And I could not decide my self to buy a VS1053 because I could not be 100 % sure.

Nowadays aac is long established, and it's the only thing I'm interested in.

Edzelf, 2021-07-05:
The ESP8266 version of the radio is replaced by the ESP32 version.  Please use the ESP32 version.  AAC is supported by the VS1053 and therefore also by the radio. If AACP is supported by the VS1053, it will probably work on the radio as well.