Edzelf/Esp-radio #186: Bluetooth module instead of VS1053

dagisr, 2020-03-30:
Could it be possible to replace VS1053 with some bluetooth module to play radio wireless using a BT-speaker. How?

Edzelf, 2020-03-30:
Without software changes you may connect a bluetooth audio transmitter to the VS1053.

dagisr, 2020-04-07:
Thank you.
How do you do that. Which BT module please

Edzelf, 2020-04-07:
Use one like [this](https://www.aliexpress.com/item/4000411307894.html?spm=a2g0o.productlist.0.0.70085a78zdiDhm&algo_pvid=1ebda2ce-fe8e-4c5d-bfa0-c994c15ed23a&algo_expid=1ebda2ce-fe8e-4c5d-bfa0-c994c15ed23a-10&btsid=0b0a187b15862516227496143ef628&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) and plug it into the VS1053.

dagisr, 2020-04-16:
yes of course! but
how to do it replacing VS1053.
in other words - feeding a BT module direct digitally from maybe esp32

Edzelf, 2020-04-16:
That's not possible with the current software.