Edzelf/Esp-radio #187: Do not auto start radio after mp3 playback

nygma2004, 2020-04-23:
Hi Edzelf,

First of all great work and thank you for sharing this. I am using your sketch as an mp3 player, not as a internet radio. So I use the option to play a URL. And it works great. But when it is done, the radio is re-started that I don't want.

What would be the best way of avoiding that?

Thanks,
Csongor

Edzelf, 2020-04-23:
And what would you do after the playing the URL?

nygma2004, 2020-04-23:
I want to use this as a simple Google Home device, I will generate some mp3-s on my server using text-to-speed and send the URL for this device to play the speech. After that I just want it to stop and do not do anything.
What I have done as a workaround is set for first station to 127.0.0.1 and deleted all other presets. So if fails to load preset 0 and stops doing anything.

Edzelf, 2020-04-23:
That's a good solution.  You may also change the software for a unlimited time-out on no data.

nygma2004, 2020-04-23:
Can you help me where do I find that?

Edzelf, 2020-04-23:
You may try to comment out line 180:

`// ini_block.newpreset++ ;                                   // Yes, try next channel`

(untested)