Edzelf/Esp-radio #188: Esp radio stops and changes channel every 25 minutes

pinkobalinko, 2020-04-24:
Hello!
Thanks for your great work on this project. And above all on support offered!
I made this project on NodeMCU v3 (esp-12F). Works like intended, except for similar situation than [here](https://github.com/Edzelf/Esp-radio/issues/110).
No matter what channel, radio stops and skips to next channel after exactly 25 minutes (1500 seconds).  Now, this may be network issue, but went unnoticed till now on other devices - radio keeps playing on other devices without interruptions.
If I manually switch channels the 25 minutes counter resets.
Could it be that built in watchdog is to sensitive? How do I disable it?

pinkobalinko, 2020-04-25:
The more I look into this, the more I think it's because of lag.
My ESP radio apparently plays a little bit slower. You can't notice while listening, it's maybe 3% slower, around 2 seconds per minute. That means, after 20 minutes, lag is 40 seconds already. And that 40 seconds of stream have to be stored in buffer, right?

Edzelf, 2020-04-25:
Strange.  Looks like the clock frequency of the VS1053 is to low.  Could you try a spare VS1053?

pinkobalinko, 2020-04-25:
Thanks for answer. I thought that, too. Tried another VS1053, same result, but spare module is from the same "bunch" and they both might be faulty.
I'll try a different one that I have. I'll report.

pinkobalinko, 2020-04-25:
Well, for now looks like I have bad luck with VS1053. The different one I can't get to work, but I noticed  that frequency written on crystal on the non working VS is slightly higher than on other two (those that play to slow) - 12.288 MHz vs. 12.000 MHz.
VS1053 datasheet states input clock frequency from 12 MHz to 13 MHz. According to [this](http://www.vsdsp-forum.com/phpbb/viewtopic.php?t=481) post, stream with 44 kHz sampling rate should play fine on 12 MHz crystal, while stream with 48 kHz sampling rate would play slower.
Looks like I have to order new VS1053. In meantime I might swap crystal from one VS1053 to another.

pinkobalinko, 2020-04-26:
OK, I managed to get to work the other VS1053 (the one with 12.288 MHz - MOSI and MISO were mislabeled) and now radio works as it should. Looks like we have to be careful when ordering VS1053s.
My question is: can I still use 12 MHz VS1053 somehow or are they only waste bin material?
Note: I edited title of the theme.

Edzelf, 2020-04-26:
The difference between 12.288 and 12.000 explains the 3 percent difference in playing speed.  Maybe you can replace the xtal with the right one.  This will cost you 10 dollarcent or so.  Chinese manufacturers tend to replace components with ones that are "just as good as the original".

pinkobalinko, 2020-04-26:
I'll certainly try that.
I'll also order some new vs1053 modules, but you can never know which you get, since click freq is usually not specified and picture is not always accurate.
Thanks again for your work and for putting me on the right track in solving this problem.

pinkobalinko, 2020-04-28:
I might have found the solution.
VS1053 datasheet states clock frequency from 12 MHz to 13 Mhz - typical is 12.288 which is right for 48 kHz sampling. I was thinking, if xtali frequency can be different but it affects playing speed, the data about that frequency must be written somewhere in VS1053 registry so player can compensate.
So I looked into [datasheet](http://www.vlsi.fi/fileadmin/datasheets/vs1003.pdf) and found that this data is written in SCI_CLOCKF registry on the address 0x3. And, luckily, there is example of calculating value for 12 MHz xtali:

> Example: If SCI_CLOCKF is 0x9BE8, SC_MULT = 4, SC_ADD = 3 and SC_FREQ = 0x3E8 = 1000.
This means that XTALI = 1000 × 4000 + 8000000 = 12 MHz. The clock multiplier is set to
3.0×XTALI = 36 MHz, and the maximum allowed multiplier that the firmware may automatically
choose to use is (3.0 + 1.5)×XTALI = 54 MHz.

So I modified line 599 from
`write_register ( SCI_CLOCKF, 6 << 12 ) ;`
to
`write_register ( SCI_CLOCKF, 0x9BE8 ) ;`

Radio works now correctly with 12 MHz VS1053. At least for mp3 with 44 kHz sampling rate.

Edzelf, 2020-04-28:
Yes, great!

pinkobalinko, 2020-04-28:
I will still try to find solution for AAC which is now not working.

Edzelf, 2020-04-28:
If you have a VS1003, it will not work.

pinkobalinko, 2020-04-28:
And it doesn't work. Well in fact I have VS1003 and all the time I thought I have VS1053 :rofl: But chip is in fact VS1003b.
And according to VS1053 datasheet, the right number for 12 MHz VS1053 is 0x8BE8.