Edzelf/Esp-radio #189: SPI fast error

drunagr, 2020-05-20:
dear Ed
i made a simple version without a screen and i got this error message, can you help?
[image]
[image]

drunagr, 2020-05-20:
when i connect the speakers i got only the FAST SPI:
[image]
this is the connections : 
[image]

h1aji, 2022-02-07:
@drunagr connect XRST pin to EN pin on esp8266 directly instead of 8k2 resistor. That diagram doesnt seem to be right