Edzelf/Esp-radio #191: Only 2 sample stations are working

zoltanszalontay, 2020-09-27:
Hi, great project!

I put together a Wemos D1 mini + VS1053 based minimal (headless) config following the 8th page of your PDF. I used your sample station presets, but only highlighted stations 0 and 7, NAXI and GAIA, are streaming. For all others, I get "Request failed" and "No data input" errors and the ESP8266 skips for the next station.

I am using the latest build. Installed yesterday. 

> preset = 6 # Start with preset 6
> **preset_00 = 109.206.96.34:8100 # 0 - NAXI LOVE RADIO, Belgrade, Serbia**
> preset_01 = airspectrum.cdnstream1.com:8114/1648_128 # 1 - Easy Hits Florida 128k
> preset_02 = us2.internet-radio.com:8050 # 2 - CLASSIC ROCK MIA WWW.SHERADIO.COM
> preset_03 = airspectrum.cdnstream1.com:8000/1261_192 # 3 - Magic Oldies Florida
> preset_04 = airspectrum.cdnstream1.com:8008/1604_128 # 4 - Magic 60s Florida 60s Classic Rock
> preset_05 = us1.internet-radio.com:8105 # 5 - Classic Rock Florida - SHE Radio
> preset_06 = icecast.omroep.nl:80/radio1-bb-mp3 # 6 - Radio 1, NL
> **preset_07 = 205.164.62.15:10032 # 7 - 1.FM - GAIA, 64k**
> preset_08 = skonto.ls.lv:8002/mp3 # 8 - Skonto 128k
> preset_09 = 94.23.66.155:8106 # 9 - *ILR CHILL and GROOVE

According to the log, all other stations send no data:

> 17:42:32.163 -> D: New preset/file requested (1/0) from us2.internet-radio.com:8050
17:42:32.201 -> D: Connect to new host us2.internet-radio.com:8050
17:42:32.201 -> D: Connect to us2.internet-radio.com on port 8050, extension /
17:42:41.486 -> D: No data input
17:42:41.486 -> D: Trying other station/file...
17:42:42.485 -> D: Request us2.internet-radio.com:8050 failed!
17:42:42.485 -> D: STOP requested
17:42:42.519 -> D: Song stopped correctly after 0 msec
17:42:43.170 -> D: New preset/file requested (2/0) from airspectrum.cdnstream1.com:8000/1261_192
17:42:43.170 -> D: Connect to new host airspectrum.cdnstream1.com:8000/1261_192
17:42:43.203 -> D: Connect to airspectrum.cdnstream1.com on port 8000, extension /1261_192
17:42:51.502 -> D: No data input
17:42:51.502 -> D: Trying other station/file...
17:42:53.469 -> D: Request airspectrum.cdnstream1.com:8000/1261_192 failed!
17:42:53.469 -> D: STOP requested
17:42:53.502 -> D: Song stopped correctly after 0 msec
17:42:54.169 -> D: New preset/file requested (3/0) from airspectrum.cdnstream1.com:8008/1604_128

Any help would be appreciated!

Thanks!
Z

Edzelf, 2020-09-27:
I do not know which stations are still up and runnning, but "icecast.omroep.nl:80/radio1-bb-mp3" should work.  You can test if a station is still up by testing it in your Internet browser or by Winamp.

zoltanszalontay, 2020-09-29:
Hi, that one and NAXI plus Gaia from the default radio.ini are working. However, all of my own URL-s fail.

Examples: Klub Radio, Hungary. 3 official URL-s. All fail. 

preset = 0 # Start with preset 0
preset_00 = stream.klubradio.hu:8080/bpstream			# 2 - Klub Rádió 96kbps
preset_01 = stream.klubradio.hu:8080/source			# 3 - Klub Rádió 96kbps
preset_02 = stream.klubradio.hu:8080/32kbit			# 4 - Klub Rádió 32kbps

Log:
10:48:43.752 -> D: New preset/file requested (0/0) from stream.klubradio.hu:8080/source
10:48:43.752 -> D: Connect to new host stream.klubradio.hu:8080/source
10:48:43.793 -> D: Connect to stream.klubradio.hu on port 8080, extension /source
10:48:54.064 -> D: Request stream.klubradio.hu:8080/source failed!
10:49:01.612 -> D: No data input

Chrome plays all of them well. Any idea?

zoltanszalontay, 2020-09-29:
I also tried a lot of other Hungarian URL-s from here. 
[https://github.com/gyarakilaszlo/Radiotray-hu/blob/master/.local/share/radiotray/bookmarks.xml](url)
Winamp gives the same URL, Chrome plays them well. Esp-radio says "No data" for these URL-s.

Edzelf, 2020-09-29:
Are you sure you have the right chip in the VS1053 module?
is "icecast.omroep.nl:80/radio1-bb-mp3" working?

zoltanszalontay, 2020-09-29:
Nope :-(
> 14:23:45.653 -> D: Request icecast.omroep.nl:80/radio1-bb-mp3 failed!
> 14:23:54.428 -> D: No data input

Again, some streams, e.g. "205.164.62.15:10032" are working.

The decoder chip is VS1053B. The PCB looks exactly like the one in your PDF, but green.

zoltanszalontay, 2020-09-29:
I just recognized that the URL you gave me is identical to the one you gave yesterday. It played yesterday, but not today. Weird.

Edzelf, 2020-09-29:
Strange.  Can you try another WiFi network in the neighbourhood?

zoltanszalontay, 2020-09-29:
Sure, I can check that in the evening. Until then, I found that 

> if ( mp3client->connect ( hostwoext.c_str(), port ) ) 

in connecttohost() returns 0, even though it splits the URL well:
> Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3

My wifi should be fine, anyway as a web browser plays this URL well, but anyhow, I'll check it with another.

zoltanszalontay, 2020-09-30:
Well, I connected to my iPhone's hotspot and all station started streaming immediately! Great find!

I just can't imagine what my Asus wifi router filters out. Firewall is enabled, but URL filter, keyword filter and network services aren't. 

Never had any problems with accessing any URL-s. And I can stream those stations from a browser on this wifi network.

zoltanszalontay, 2020-10-01:
A got a bit further. DNS name resolution is not working when using my home wifi network. Only stations with IP addresses, like NAXI or GAIA radio work. 

However, when connecting to my iPhone's hotspot, name resolution is working.

About my config: DHCP Server is my wifi router. DNS is set to the IP of my Synology NAS, which forwards all requests to 8.8.8.8 (Google). When I use iPhone hotspot, the DNS is external (172.20.10.1).

I suspect this problem is related to LWIP's DNS forwarding, but still investigating.

Edzelf, 2020-10-02:
Glad you found the cause.  Maybe the DNS request takes to long going through the NAS software (ad blocking?).

zoltanszalontay, 2020-10-02:
I'll check the timestamp of the name resolution. Also, I am going to set 8.8.8.8 in my DHCP scope. I'll lose the ability to access my local computers by names as I do not use WINS, but that's not a problem. No ad blocking takes place on the server side. My last tip would have been WIFI_STA mode, but that's being used.

Until then, I use IP addresses.

Great piece of software! Thanks!

P.S.: You may close this issue as it is up to the IP stack or my network config. Thx

closed 2020-10-02 by Edzelf

zoltanszalontay, 2020-10-02:
Got it!

Setting a public DNS Server (e.g. 8.8.8.8) on my router's DHCP scope solved the problem so it was the DNS forwarding initiated by my Synology NAS.