Edzelf/Esp-radio #192: Connect to internet radio stream - Radio Nowy Świat

macwojs, 2020-10-12:
Hello
I tried to connect to this radio station: https://nowyswiat.online/gdzie-nas-slychac/
I tried with http and https connection:

- http - n06a-eu.rcs.revma.com:443/ypqt40u0x1zuv

> 01:12:06.847 -> D: Command: station with parameter stream.rcs.revma.com:80/ypqt40u0x1zuv
> 01:12:06.847 -> D: Connect to new host stream.rcs.revma.com:80/ypqt40u0x1zuv
> 01:12:06.914 -> D: Connect to stream.rcs.revma.com on port 80, extension /ypqt40u0x1zuv
> 01:12:07.146 -> D: Connected to server
> 01:12:07.179 -> D: Switch to HEADER
> 01:12:07.179 -> D: Location: http://n12a-eu.rcs.revma.com/ypqt40u0x1zuv?rj-ttl=5&rj-tok=AAABdR8fAZ0AzdwDEsVInTelFQ
> 01:12:07.179 -> D: Content-Length: 0
> 01:12:07.179 -> D: Connection: close
> 01:12:07.179 -> D: Set-Cookie: rj-listener-cookie=13fh7i0cwweth; Domain=revma.com
> 01:12:18.670 -> D: No data input
> 01:12:27.449 -> D: Command: status (without parameter)
> 01:12:28.678 -> D: No data input
> 01:12:28.678 -> D: Trying other station/file...

- https - stream.nowyswiat.online:443/mp3

> 01:14:11.146 -> D: Command: station with parameter stream.nowyswiat.online:443/mp3
> 01:14:11.610 -> D: Connect to new host stream.nowyswiat.online:443/mp3
> 01:14:11.676 -> D: Connect to stream.nowyswiat.online on port 443, extension /mp3
> 01:14:11.875 -> D: Connected to server
> 01:14:11.908 -> D: Switch to HEADER
> 01:14:11.908 -> D: Server: cloudflare
> 01:14:11.908 -> D: Content-Type: text/html
> 01:14:11.908 -> D: text/html seen.
> 01:14:11.908 -> D: Content-Length: 253
> 01:14:11.908 -> D: Connection: close
> 01:14:11.908 -> D: CF-RAY: -
> 01:14:11.908 -> D: Switch to DATA, bitrate is 0, metaint is 0
> 01:14:11.908 -> D: First chunk:
> 01:14:11.908 -> D: 3C 68 74 6D 6C 3E 0D 0A
> 01:14:11.908 -> D: 3C 68 65 61 64 3E 3C 74
> 01:14:11.908 -> D: 69 74 6C 65 3E 34 30 30
> 01:14:11.941 -> D: 20 54 68 65 20 70 6C 61
> 01:14:28.666 -> D: No data input
> 01:14:38.662 -> D: No data input
> 01:14:38.662 -> D: Trying other station/file...

I must use another url or what?

I use D1 mini with esp8266 and preset radio station mostly work fine.

Edzelf, 2020-10-13:
Try: "stream.rcs.revma.com/ypqt40u0x1zuv" without "http://"
Request to stream.nowyswiat.online/mp3 are redirected to:
Location: https://stream.rcs.revma.com/ypqt40u0x1zuv
HTTPS is not supported by the radio, but since http://stream.rcs.revma.com/ypqt40u0x1zuv also works, you can use
"stream.rcs.revma.com/ypqt40u0x1zuv"

macwojs, 2020-10-13:
I tried and still can't connect to radio:

> 11:07:11.747 -> D: Connect to new host stream.rcs.revma.com/ypqt40u0x1zuv
> 11:07:11.813 -> D: Connect to stream.rcs.revma.com on port 80, extension /ypqt40u0x1zuv
> 11:07:12.112 -> D: Connected to server
> 11:07:12.179 -> D: Switch to HEADER
> 11:07:12.179 -> D: Location: http://n08a-eu.rcs.revma.com/ypqt40u0x1zuv?rj-ttl=5&rj-tok=AAABdSE_0lQAkezm-wizn7WJ1Q
> 11:07:12.212 -> D: Content-Length: 0
> 11:07:12.212 -> D: Connection: close
> 11:07:12.212 -> D: Set-Cookie: rj-listener-cookie=14il1a2q580j9; Domain=revma.com
> 11:07:13.308 -> D: No data input
> 11:07:13.308 -> D: Trying other station/file...

Maybe it's problem with ESP3266 version as here: https://github.com/Edzelf/Esp-radio/issues/183. You use ESP32? If this is the issue then I need to buy ESP32.

Edzelf, 2020-10-13:
Maybe the key "ypqt40u0x1zuv" changes for every session.  AT this moment it still works for me.
Try to get the right location (in the logging) by connecting to "stream.nowyswiat.online/mp3"

macwojs, 2020-10-13:
The same:

> 11:57:12.105 -> D: Command: station with parameter stream.nowyswiat.online/mp3
> 11:57:12.138 -> D: Connect to new host stream.nowyswiat.online/mp3
> 11:57:12.171 -> D: Connect to stream.nowyswiat.online on port 80, extension /mp3
> 11:57:12.371 -> D: Connected to server
> 11:57:12.404 -> D: Switch to HEADER
> 11:57:12.404 -> D: Content-Type: text/html
> 11:57:12.404 -> D: text/html seen.
> 11:57:12.404 -> D: Transfer-Encoding: chunked
> 11:57:12.404 -> D: Connection: close
> 11:57:12.404 -> D: Set-Cookie: __cfduid=d979775b8a1f7c8101025d521e1e0d2b61602583032; expires=Thu, 12-Nov-20 09:57:12 G
> 11:57:12.437 -> D: Cache-Control: no-cache, no-store, must-revalidate, max-age=0
> 11:57:12.437 -> D: Location: https://stream.rcs.revma.com/ypqt40u0x1zuv
> 11:57:12.437 -> D: X-Turbo-Charged-By: LiteSpeed
> 11:57:12.437 -> D: CF-Cache-Status: DYNAMIC
> 11:57:12.437 -> D: cf-request-id: 05c2fd8a2a0000f290f013e200000001
> 11:57:12.437 -> D: Report-To: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report?lkg-colo=73&lkg-time=160258
> 11:57:12.437 -> D: Server: cloudflare
> 11:57:12.470 -> D: CF-RAY: 5e1831f049eaf290-WAW
> 11:57:12.470 -> D: Switch to DATA, bitrate is 0, metaint is 0
> 11:57:12.470 -> D: First chunk:
> 11:57:12.470 -> D: 3C 21 44 4F 43 54 59 50
> 11:57:12.470 -> D: 45 20 68 74 6D 6C 3E 0A
> 11:57:12.470 -> D: 3C 68 74 6D 6C 20 73 74
> 11:57:12.470 -> D: 79 6C 65 3D 22 68 65 69
> 11:57:13.466 -> D: No data input
> 11:57:13.466 -> D: Trying other station/file...

Edzelf, 2020-10-13:
I tested it on the ESP32 version of the radio.  I am not sure if chunked transfer is supported on the ESP8266 version.

macwojs, 2020-10-13:
Okej, so I'm waiting for my esp32 and then I get it I will send update about this issue

Edzelf, 2020-10-13:
I just test this on my ESP8266 radio.  Same result: connaction oka, but chunked stream can not be decoded.

closed 2020-10-19 by macwojs

Dubik12, 2021-02-17:
> Maybe the key "ypqt40u0x1zuv" changes for every session. AT this moment it still works for me.
> Try to get the right location (in the logging) by connecting to "stream.nowyswiat.online/mp3"

Not this key does not change. The end of the address is changed, and the server number is changed (at the beginning of the HTTP address). For example:
http://n14a-eu.rcs.revma.com/ypqt40u0x1zuv?rj-ttl=5&rj-tok=AAABd7E0lvsAjObIsY8Nc-MKtA
 I try to use the command on your Radio:
//-----------------------------------------------
const char* headerNames[] = { "Location" };

    String Location;

HTTPClient http;
http.begin(URL);
    http.collectHeaders(headerNames, sizeof(headerNames)/sizeof(headerNames[0]));

    int httpCode = http.GET();  
Serial.print (httpCode);
 if (httpCode == 302)
    {
     Serial.println("following redirect"); 
    Location = http.header("Location"); 
 Serial.print ("LOCATION   "); Serial.println (Location);    
 char url[100];    
int  pos = Location.length();   
 Serial.print("Location.length()  ");    Serial.println (pos);
  Location.substring(0, pos).toCharArray(url, pos+1);   
  Serial.println (url);
     }

//---------------------------------------------
They are located after WiFi "connect lines" & 
" file = new AudioFileSourceICYStream(url);
  file->RegisterMetadataCB(MDCallback, (void*)"ICY");"
The answer is a valid link, but it has already expired. or maybe a future link. 
Can You help? This is the only way to listen to this radio