Edzelf/Esp-radio #193: Synchronized / synchronous playback on multiple ESP radios?

Wookbert, 2021-01-15:
NOOB question: I want to listen to the same internet radio stations in several rooms of my apartment (bath, kitchen, dining room etc.). I see two possible paths:

**Path A – A single ESP radio receiver with multiple Bluetooth 5.0 transmitters** attached to the analogue audio output of the VS1053. BT range is a certain issue here.
What I don't like in this setup is the fact that the audio signal goes Digital->Analogue->Digital->Analogue  (ESP->VS1053->BT TX->BT RX).

**Path B – Several parallel ESP radios**, one in each room.

The big question regarding path B: Is it possible to have the four or five ESP radios play totally synchronous, using e.g. an internet time server and some sort of time code/stamp? If yes, how exactly.

Thanks in advance!

Edzelf, 2021-01-15:
I don't think it is possible to synchronize two internet radios.

Wookbert, 2021-01-15:
I guess I'm not alone when it comes to the need for affordable (wireless) multi-room radio.

A totally naive idea/suggestion: Couldn't you – time and interest given – develop an ESP Radio Server which receives the stream, adds some sort of time code on top and rebroadcasts to ESP Radio Clients, whereas Server and Clients have synchronized clocks?

Perhaps one can inspect or implement code from other [multi-room projects](https://github.com/topics/multiroom-audio), e.g. [SnapCast](https://github.com/badaix/snapcast).

Having a single ESP32-based server and multiple, super low cost  EPS8266-based clients is really appealing.