Edzelf/Esp-radio #196: Adafruit_I2CDevice.h and Adafruit_SPIDevice.h missing

newbee90, 2021-01-24:
hello, 
I have Arduiono1.8.13 installed as a portable version, to have a clean version
and esp8266 community vers 2.2.0
and all the libraries I found in Esp_radio.ino
now I get this error message
libraries\Adafruit-GFX-Library\Adafruit_GrayOLED.h:30:32: fatal error: Adafruit_I2CDevice.h: No such file or directory
and after if I cange the entry in Adafruit_GrayOLED.h to //#include <Adafruit_I2CDevice.h>
libraries\Adafruit-GFX-Library\Adafruit_GrayOLED.h:31:32: fatal error: Adafruit_SPIDevice.h: No such file or directory
is there a Esp_radio.bin which I could flash with NodeMCU-PyFlasher?

Edzelf, 2021-01-24:
I have uploaded the sketch in binary format to the bin directory.

newbee90, 2021-01-25:
I have no sound, I can see Artist and Title information at the webside, so the radio work
I have checked the wiring several times, but all is correct
is your bin file with tft or without tft?

Edzelf, 2021-01-25:
Show the logging at the serial port.

newbee90, 2021-01-25:
23:55:18.081 -> D: Slow SPI,Testing VS1053 read/write registers...
23:55:18.081 -> D: VS1053 error retry SB:012C R1:0000 R2:0000
23:55:18.081 -> D: VS1053 error retry SB:0258 R1:0000 R2:0000
23:55:18.128 -> D: VS1053 error retry SB:0384 R1:0000 R2:0000
23:55:18.128 -> D: VS1053 error retry SB:04B0 R1:0000 R2:0000
23:55:18.128 -> D: VS1053 error retry SB:05DC R1:0000 R2:0000
23:55:18.181 -> D: VS1053 error retry SB:0708 R1:0000 R2:0000
23:55:18.181 -> D: VS1053 error retry SB:0834 R1:0000 R2:0000
23:55:18.181 -> D: VS1053 error retry SB:0960 R1:0000 R2:0000
23:55:18.181 -> D: VS1053 error retry SB:0A8C R1:0000 R2:0000
23:55:18.228 -> D: VS1053 error retry SB:0BB8 R1:0000 R2:0000
23:55:18.228 -> D: VS1053 error retry SB:0CE4 R1:0000 R2:0000
23:55:18.228 -> D: VS1053 error retry SB:0E10 R1:0000 R2:0000
23:55:18.281 -> D: VS1053 error retry SB:0F3C R1:0000 R2:0000
23:55:18.281 -> D: VS1053 error retry SB:1068 R1:0000 R2:0000
23:55:18.281 -> D: VS1053 error retry SB:1194 R1:0000 R2:0000
23:55:18.328 -> D: VS1053 error retry SB:12C0 R1:0000 R2:0000
23:55:18.328 -> D: VS1053 error retry SB:13EC R1:0000 R2:0000
23:55:18.328 -> D: VS1053 error retry SB:1518 R1:0000 R2:0000
23:55:18.381 -> D: VS1053 error retry SB:1644 R1:0000 R2:0000
23:55:18.381 -> D: VS1053 error retry SB:1770 R1:0000 R2:0000
23:55:18.482 -> D: Fast SPI, Testing VS1053 read/write registers again...
23:55:18.529 -> D: VS1053 error retry SB:0003 R1:0000 R2:0000
23:55:18.529 -> D: VS1053 error retry SB:0006 R1:0000 R2:0000
23:55:18.529 -> D: VS1053 error retry SB:0009 R1:0000 R2:0000
23:55:18.582 -> D: VS1053 error retry SB:000C R1:0000 R2:0000
23:55:18.582 -> D: VS1053 error retry SB:000F R1:0000 R2:0000
23:55:18.582 -> D: VS1053 error retry SB:0012 R1:0000 R2:0000
23:55:18.582 -> D: VS1053 error retry SB:0015 R1:0000 R2:0000
23:55:18.629 -> D: VS1053 error retry SB:0018 R1:0000 R2:0000
23:55:18.629 -> D: VS1053 error retry SB:001B R1:0000 R2:0000
23:55:18.629 -> D: VS1053 error retry SB:001E R1:0000 R2:0000
23:55:18.682 -> D: VS1053 error retry SB:0021 R1:0000 R2:0000
23:55:18.682 -> D: VS1053 error retry SB:0024 R1:0000 R2:0000
23:55:18.682 -> D: VS1053 error retry SB:0027 R1:0000 R2:0000
23:55:18.729 -> D: VS1053 error retry SB:002A R1:0000 R2:0000
23:55:18.729 -> D: VS1053 error retry SB:002D R1:0000 R2:0000
23:55:18.729 -> D: VS1053 error retry SB:0030 R1:0000 R2:0000
23:55:18.783 -> D: VS1053 error retry SB:0033 R1:0000 R2:0000
23:55:18.783 -> D: VS1053 error retry SB:0036 R1:0000 R2:0000
23:55:18.783 -> D: VS1053 error retry SB:0039 R1:0000 R2:0000
23:55:18.783 -> D: VS1053 error retry SB:003C R1:0000 R2:0000
23:55:18.829 -> D: endFillByte is 0
23:55:19.986 -> D: Selected network: FRITZBox7490             
23:55:19.986 -> D: Try WiFi FRITZBox7490
23:55:23.072 -> D: IP = 192.168.178.41
23:55:23.125 -> D: Start server for commands
23:55:23.125 -> D: Connecting to MQTT broker.hivemq.com, port 1883, user none, password none...
23:55:23.373 -> D: MQTT Connected to the broker broker.hivemq.com, session is not present
23:55:23.373 -> D: Subscribing to espradio at QoS 2, packetId = 1 
23:55:23.373 -> D: Publishing IP 192.168.178.41 to topic espradioIP
23:55:23.426 -> D: MQTT Subscribe acknowledged, packetId = 1, QoS = 2
23:55:23.426 -> D: MQTT Publish acknowledged, packetId = 2
23:55:24.128 -> D: BASS settings is 0000
23:55:24.128 -> D: STOP requested
23:55:24.175 -> D: Song stopped correctly after 0 msec
23:55:24.830 -> D: New preset/file requested (-1/0) from 
23:55:24.877 -> D: New preset/file requested (-1/0) from webstream.radio-bamberg.de/radio-bamberg.mp3
23:55:24.877 -> D: Connect to new host webstream.radio-bamberg.de/radio-bamberg.mp3
23:55:24.877 -> D: Connect to webstream.radio-bamberg.de on port 80, extension /radio-bamberg.mp3
23:55:25.023 -> D: Connected to server
23:55:25.209 -> D: Switch to HEADER

Edzelf, 2021-01-25:
All those VS1053 errors shows that there is a (wiring) problem withe the SPI bus.

newbee90, 2021-01-25:
https://ibb.co/R7fcCDG
https://ibb.co/b2T91t9
I made it new, same problem

Edzelf, 2021-01-26:
Wiring looks good.  Check if the chip on the MP3 board is really an VS1053.
Check if the reset pin of the VS1053 board is high.
It should also work without the diodes.

newbee90, 2021-01-26:
now I have remove the diodes and connected reset to vss, now I have sound but with many breaks, like very old CD
https://ibb.co/wsKfR8Y it is an VS1053