Edzelf/Esp-radio #197: Split content with 2.42 OLED Display

martix90, 2021-02-16:
Hi, 
I really like this very cool project and want to build my own WebRadio.

I would like to use a 2.42 OLED display that I have around. This uses an SSD1309 Chip, which (according to most Docu) is almost identical to the SSD1306. But when I configured this project to use the OLED display, the image is split in half and the left side is right and the right side is left (see image). 
[image]

If I use an example sketch with u8g2 the display works just fine.

Is there any idea what could cause this issue? Or is this type of display simply not supported? I think there's not much missing for it to work correctly...

Edzelf, 2021-02-17:
You need to modify the constructor and the init call on line 23 and 36 of "oled.h". It is now:
U8G2_SSD1306_128X64_NONAME_1_SW_I2C
And it should be the one for 1309.  See [https://github.com/olikraus/u8g2/wiki/u8g2setupcpp](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp).
For example:
U8G2_SSD1309_128X64_NONAME2_1_SW_I2C

martix90, 2021-02-17:
Hi @Edzelf, first thank you very much for your answer. I just noticed I'm in the wrong project with my question (I'm using an ESP32) - but I think you already noticed - reading your answer :)

To be honest, I don't quite understand it yet. If "OLED" is defined, the file SSD1306.h is included. Here, oled.h is never used. If I instead include oled.h (where I have changed the two lines of your comment), it doesn't compile since oled.h does not have all the required functions (like for example displayvolume() ). 
I think I did not understand something correct here :(

Edzelf, 2021-02-17:
You are right.
If you uncomment the #define OLEd in line 178 o main.cpp and comment the line with BLUETFT.
Then SSD1306.h is included.
The file oled.h is not in use anymore.
The driver for the SSD1306 is not quite compatible with the 1309.
I will try to fix that later.

martix90, 2021-02-17:
Thank you very much!

Edzelf, 2021-02-19:
Try the new version.  I did not test it, because I don't have the right module.
3 models are available:
```
//#define OLED1306                     // 64x128 I2C OLED SSD1306
//#define OLED1309                     // 64x128 I2C OLED SSD1309
//#define OLED1106                     // 64x128 I2C OLED SH1106
```

martix90, 2021-02-20:
Wow - I'm really impressed with how fast you did the refactoring of the oled drivers. I will try it as soon as possible :) Thank you so much!

martix90, 2021-02-20:
Hi @Edzelf, I've just tested it. It's a lot better but not yet quite right. The lines seem to start at different columns Maybe an off-by-one error?

[image]

Edzelf, 2021-02-20:
Did you try all 3 models?  Please give me the exact type of the display.  Maybe I can find a hint in the datasheet.Vriendelijke groeten,Ed.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Edzelf/Esp-radio/issues/197#issuecomment-782598065",
"url": "https://github.com/Edzelf/Esp-radio/issues/197#issuecomment-782598065",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

martix90, 2021-02-20:
Hi Ed, I now tried all three models (OLED1306, OLED1309 and OLED1106).
All of them behave exactly the same.

The model I use is called "DIY MORE" and says on the back "2,42' OLED, SSD1309", here's a links:
https://www.diymore.cc/products/2-42-inch-12864-oled-display-module-iic-i2c-spi-serial-for-arduino-c51-stm32-green-white-blue-yellow

Thank you and greetings from Germany
Martin

martix90, 2021-02-24:
I think I found the issue:
If I comment out  this line, it works :)

//i2c_master_write ( i2Cchan, fillbuf, 4, true ) ;      // Yes, fill extra RAM with zeroes

Edzelf, 2021-02-24:
Great!