Edzelf/Esp-radio #198: Ols SSID in list

ilioss, 2021-02-28:
Hi Ed,

Long time no see.
Last half a year I moved to un other house.
So the old SSID has to be changed.
My new router is Ziggo connection box.
If I changed the old SSID name and password in the pref. file.
It will not connect to the wifi???
The serial monitor still mention that the old SSID is added to the list of networks.
Connect to WiFi     ( the new SSID/password in the pref.file)
WiFi failed!  Trying .....
 
The resent not connect issue ZI tried but no result.
BTW in the PlatformIO the project is compiled and uploaded ( proof by turning the display ) must be corrected.
But it also complained that the include is missing?
Any cloe about this issues.
I preferably like to use the radio again while Ziggo is hopeless with his radio.
And I will use platformIO
Like to hear.
Kind regards Theo

Edzelf, 2021-02-28:
Try to connect to the wifi access point that is available if the radio cannot connect to the network.  Then goto 192.168.4.1 and add your ziggo credentials to the config page of the web interface.

ilioss, 2021-02-28:
Hallo Ed,Top bedankt.Werkt weer als een jekko.Btw om mijn opzet wat degelijker te maken een nakedninje pcb besteld.Hoewel de radio zoals hij is door de jaren heen weer iets heeft van waarom veranderen.Nog steeds erg onder de indruk.Met vriendelijke groet,Theo