Edzelf/Esp-radio #199: Has anyone uploaded the sketch wirelessly using OTA? Is you did, how did you do it? I can;t get to work.

wa8lbz, 2021-03-02:


Edzelf, 2021-03-02:
First you have to upload a sketch that is able to handle OTA.  Then you can specilfy OTA as upload method to upload a new sketch that can also handle OTA.

Derek-K, 2021-07-02:
Assume you are using Arduino IDE.

The very first time, you need to flash via USB (unless you already have OTA firmware installed on your ESP8266).

From that point on, assume your computer is on the same network as your ESP-Radio. Under Arduino IDE, go to Tools -> Port: xxxxx  Select ESP-radio at xxx.xxx.xxx.xxx. (Note, serial monitor no longer works. Use another terminal software instead, e.g. Putty). upload the code as usual, it will upload Over-the-Air.