Edzelf/Esp-radio #203: Newbie needing help with TFT colours and fonts

rjmusto, 2021-05-12:
Hi all,
Firstly a big thanks to Ed for sharing all his hard work on this Internet radio. Amazing stuff.

I have got mine up and running and am using an ST7735 1.8" display, and am very happy to have got it working.

As there is plenty of room on the display, I would really like to make the font bigger and maybe add some background colours, but can't work out how to do this (not new to Arduino code but certainly no expert...). I've seen the;

"#define dsp_setTextSize(a)      tft->setTextSize(a)                // Set the text size"

line in the bluetft.h tab, but am not sure what to do with it.   I've read about setTextSize(1) and setTextSize(2), but no idea how the (a) in this example relates to that.

And as for background colours.... no idea where to start.

Any help much appreciated.

Thanks,
Ralph

Edzelf, 2021-05-13:
dsp_setText is just a macro.  The goal is to make a function "dsp_setText()" that will work on all included display drivers.
So if you want to change the size, look for calls of "dsp_setText" in the main program.
For colours, look for "BLACK" or "RED" or "YELLOW" in the main program.

rjmusto, 2021-05-15:
Hi Ed,
Many thanks for the reply, much appreciated.

My main interest was to make the text in the middle block that displays the station being played a bit bigger - helps with my ageing eysight.....

Thanks for the hints - I am gradually developing a better understanding of the sketch structure, so will work away on that.

BR,
Ralph

Derek-K, 2021-07-02:
I have successfully replaced the original AdadruitGFX & ILI9163C library with TFT_eSPI library with minimal modification and eventually redesigned the GUI.

Using TFT_eSPI will allow you to use many types of LCD (e.g. I am currently using a 320x240 ILI9341 with touch screen control)

h1aji, 2022-02-07:
@Derek-K would you be able to share your code with TFT_eSPI? thanks

Derek-K, 2022-03-21:
@h1aji As far as I remember, just load the TFT_eSPI library, and if there are any changes, it will be very minor.

h1aji, 2022-03-24:
@Derek-K Figured out, easy. Thanks