Edzelf/Esp-radio #210: Setting up the 128x160 display

samand587, 2021-11-04:
Dear author! This is the image the display displays. The display applied the recommended one in the project.
How can this be fixed? Thanks!   
[image]

Edzelf, 2021-11-04:
See the document.  Search for "TFT library patches".

samand587, 2021-11-04:
Made changes to TFT_ILI9163C.cpp , nothing has changed- the screen is upside down, on the left there is a white stripe for a quarter of the screen. What else can I see?
[image]

samand587, 2021-11-04:
I won the white stripe. How can I rotate the screen?

samand587, 2021-11-04:
Line 1825: 
tft.setRotation ( 1 ) Excellent solution! The question is closed. Thanks!

closed 2021-11-04 by samand587