Edzelf/Esp-radio #213: Stations with redirect will not work when SPIRAM enabled

h1aji, 2022-02-27:
Hi Ed,
I noticed that if SPIRAM enabled then stations with redirect wont work. Here is output

```
D: New preset/file requested (1/0) from n0d.radiojar.com/hcrb063nn3quv
D: Connect to new host n0d.radiojar.com/hcrb063nn3quv
D: Connect to n0d.radiojar.com on port 80, extension /hcrb063nn3quv
D: Connected to server
D: Switch to HEADER
D: Access-Control-Allow-Origin: *
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested
D: New preset/file requested (8/0) from stream.live.vc.bbcmedia.co.uk/bbc_world_service
```
Although ringbuf works perfectly fine

Edzelf, 2022-02-27:
Yes, I know.  That is because there is a delay (a number of input bytes).  The redirect info is less than that number and is therefore not handled.

h1aji, 2022-02-27:
Is there a quick way to fix it? Thanks

Also please see my version **spiram.cpp** without external library.

```
//******************************************************************************************
// SPI RAM routines.                                                                       *
//******************************************************************************************
// Use SPI RAM as a circular buffer with chunks of 32 bytes.                               *
//******************************************************************************************

#include <SPI.h>

#define SRAM_SIZE  131072                   // Total size SPI ram in bytes
#define CHUNKSIZE      32                   // Chunk size
#define SRAM_CH_SIZE 4096                   // Total size SPI ram in chunks

#define SRAM_CS        10                   // GPIO1O SRAM CS pin
#define SRAM_FREQ    16e6                   // The 23LC1024 supports theorically up to 20MHz

// SRAM opcodes
#define SRAM_READ    0x03
#define SRAM_WRITE   0x02

// Global variables
uint16_t   chcount ;                       // Number of chunks currently in buffer
uint32_t   readinx ;                       // Read index
uint32_t   writeinx ;                      // write index

uint32_t spiTransfer32 ( uint32_t data )
{
  union { uint32_t val; struct { uint16_t lsb; uint16_t msb; }; } in, out;
  in.val = data;
  out.msb = SPI.transfer16 ( in.msb ) ;
  out.lsb = SPI.transfer16 ( in.lsb ) ;
  return out.val;
}

void spiramWrite ( uint32_t addr, uint8_t *buff, uint32_t size )
{
  int i = 0;
  while ( size-- )
  {
      SPI.beginTransaction ( SPISettings ( SRAM_FREQ, MSBFIRST, SPI_MODE0 ) ) ;
      digitalWrite ( SRAM_CS, LOW ) ;
      spiTransfer32 ( (SRAM_WRITE<<24)|(addr++&0x00ffffff) ) ;   // Set write mode
      SPI.transfer ( buff[i++] ) ;
      digitalWrite ( SRAM_CS, HIGH ) ;
      SPI.endTransaction();  
  }
}

void spiramRead ( uint32_t addr, uint8_t *buff, uint32_t size )
{
  int i = 0;
  while ( size-- )
  {
      SPI.beginTransaction ( SPISettings ( SRAM_FREQ, MSBFIRST, SPI_MODE0 ) ) ;
      digitalWrite ( SRAM_CS, LOW ) ;
      spiTransfer32 ( (SRAM_READ<<24)|(addr++&0x00ffffff) ) ;   // Set read mode
      buff[i++] = SPI.transfer ( 0x00 );
      digitalWrite ( SRAM_CS, HIGH ) ;
      SPI.endTransaction();  
  }
}

//******************************************************************************************
//                              S P A C E A V A I L A B L E                                *
//******************************************************************************************
// Returns true if bufferspace is available.                                               *
[... 63 lines cut ...]
//                                S P I R A M S E T U P                                    *
//******************************************************************************************
void spiramSetup()
{
  SPI.begin();
  pinMode ( SRAM_CS, OUTPUT ) ;
  digitalWrite ( SRAM_CS, HIGH ) ;

  digitalWrite ( SRAM_CS, HIGH ) ;
  delay ( 50 ) ;
  digitalWrite ( SRAM_CS, LOW ) ;
  delay ( 50 ) ;
  digitalWrite ( SRAM_CS, HIGH ) ;

  SPI.beginTransaction ( SPISettings ( SRAM_FREQ, MSBFIRST, SPI_MODE0 ) ) ;
  digitalWrite ( SRAM_CS, LOW ) ;
  SPI.transfer ( 0x01 ) ;                             // Write mode register
  SPI.transfer ( 0x00 ) ;                             // Set byte mode
  digitalWrite ( SRAM_CS, HIGH ) ;
  SPI.endTransaction();

  bufferReset() ;                                     // Reset ringbuffer administration
}

```

Edzelf, 2022-02-27:
I have no way to test it.

h1aji, 2022-02-27:
I tested it and it works fine

Edzelf, 2022-02-27:
so, what is the problem?

h1aji, 2022-02-27:
No issue with the code itself. Just redirect is not working

h1aji, 2022-02-28:
@Edzelf I removed code with spiramdelay but it had no effect. Same out as I had above.
I even connected SPI RAM CS pin to GPIO 15 and set frequency to 20MHz, still ...

Edzelf, 2022-02-28:
Yes, I know. That is because there is a delay (a number of input bytes). The redirect info is less than that number and is therefore not handled

h1aji, 2022-03-06:
Hi @Edzelf 
I was trying to modify 
https://github.com/Edzelf/Esp-radio/blob/65f20ce4a5334ed1a272d046e8ed7f6b73dbb5e5/Esp_radio.ino#L2478

with no success.

Would you be able to advise how it can be done, maybe you can give me a clue? Thanks

h1aji, 2024-06-05:
Hi @Edzelf 
Is there any chance to make redirect working while SPIRAM option is enabled?
Thanks