Edzelf/Esp-radio #220: The board keeps restarting, instead of the website only "sorry" appears

JureqB, 2023-12-22:
I am unable to access the website because the board keeps restarting. I tried different ESP32 modules, tested with and without a connected display and decoder - the effect is the same:

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (516) esp_core_dump_flash: No core dump f�ѥѥ���found!
E (516) esp_core_dump_flash: No core dump partition found!
[     8][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3025][I][main.cpp:2582] setup(): [main] Version Thu, 14 Dec 2023 09:00:00 GMT.  Free memory 110580
[  3027][I][main.cpp:2583] setup(): [main] Display type is ILI9341
[  3123][I][main.cpp:2593] setup(): [main] SPIFFS is okay, space 1016801, used 0
[  3196][I][main.cpp:2614] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3196][I][main.cpp:2614] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3204][I][main.cpp:2614] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1114112
[  3216][I][main.cpp:2486] fillkeylist(): [main] Read 0 keys from NVS
[  3220][I][main.cpp:1910] readIOprefs(): [main] pin_spi_sck set to 18
[  3226][I][main.cpp:1910] readIOprefs(): [main] pin_spi_miso set to 19
[  3232][I][main.cpp:1910] readIOprefs(): [main] pin_spi_mosi set to 23
[  3249][I][main.cpp:2655] setup(): [main] GPIO0 is HIGH
[  3259][I][main.cpp:2655] setup(): [main] GPIO2 is HIGH
[  3269][I][main.cpp:2655] setup(): [main] GPIO4 is HIGH
[  3279][I][main.cpp:2655] setup(): [main] GPIO5 is HIGH
[  3289][I][main.cpp:2655] setup(): [main] GPIO12 is HIGH
[  3299][I][main.cpp:2655] setup(): [main] GPIO13 is HIGH
[  3309][I][main.cpp:2655] setup(): [main] GPIO14 is HIGH
[  3319][I][main.cpp:2655] setup(): [main] GPIO15 is HIGH
[  3329][I][main.cpp:2655] setup(): [main] GPIO16 is HIGH
[  3339][I][main.cpp:2655] setup(): [main] GPIO17 is HIGH
[  3349][I][main.cpp:2655] setup(): [main] GPIO18 is HIGH
[  3359][I][main.cpp:2655] setup(): [main] GPIO19 is HIGH
[  3369][I][main.cpp:2655] setup(): [main] GPIO21 is HIGH
[  3379][I][main.cpp:2655] setup(): [main] GPIO22 is HIGH
[  3389][I][main.cpp:2655] setup(): [main] GPIO23 is HIGH
[  3399][I][main.cpp:2655] setup(): [main] GPIO25 is HIGH
[  3409][I][main.cpp:2655] setup(): [main] GPIO26 is HIGH
[  3419][I][main.cpp:2655] setup(): [main] GPIO27 is HIGH
[  3429][I][main.cpp:2655] setup(): [main] GPIO32 is HIGH
[  3439][I][main.cpp:2655] setup(): [main] GPIO33 is HIGH
[  3449][I][main.cpp:2655] setup(): [main] GPIO34 is LOW, probably no PULL-UP
[  3459][I][main.cpp:2655] setup(): [main] GPIO35 is HIGH
[  3469][I][main.cpp:2655] setup(): [main] GPIO39 is LOW, probably no PULL-UP
[  3471][I][main.cpp:2672] setup(): [main] Start ILI9341 display
[  3472][E][main.cpp:2689] setup(): [main] Display not activated
[  3476][I][main.cpp:2278] mk_lsan(): [main] Create list with acceptable WiFi networks
[  3484][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: siatka
[  3491][I][main.cpp:2308] mk_lsan(): [main] Added siatka to list of networks
[  4014][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  4112][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[  4112][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[  4611][I][main.cpp:865] listNetworks(): [main] Scan Networks
[ 10826][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 7
[ 10826][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 10831][I][main.cpp:867] listNetworks(): [main] Scan completed
[ 10836][I][main.cpp:875] listNetworks(): [main] Number of available networks: 7
[ 10843][I][main.cpp:893] listNetworks(): [main]  1 - ArabicaGuest              Signal: -56 dBm, Encryption WPA_WPA2_PSK, Not accepttable
[ 10855][I][main.cpp:893] listNetworks(): [main]  2 - ARABICA2GHZ               Signal: -56 dBm, Encryption WPA2_PSK, Not accepttable
[ 10867][I][main.cpp:893] listNetworks(): [main]  3 - G6_5308                   Signal: -72 dBm, Encryption WPA2_PSK, Not accepttable
[ 10879][I][main.cpp:893] listNetworks(): [main]  4 - sieciunia                 Signal: -77 dBm, Encryption WPA2_PSK, Not accepttable
[ 10891][I][main.cpp:893] listNetworks(): [main]  5 - c96MQcTF                  Signal: -77 dBm, Encryption WPA2_PSK, Not accepttable
[ 10902][I][main.cpp:893] listNetworks(): [main]  6 - siatka                    Signal: -79 dBm, Encryption WPA2_PSK, Acceptable
[ 10914][I][main.cpp:893] listNetworks(): [main]  7 - NETIASPOT-87EF10          Signal: -82 dBm, Encryption WPA_WPA2_PSK, Not accepttable
[ 10926][I][main.cpp:897] listNetworks(): [main] End of list
[ 10931][I][main.cpp:2717] setup(): [main] Connect to network
[ 10938][V][WiFiGeneric.cpp:343] _arduino_event_cb(): STA Stopped
[ 10942][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 3 - STA_STOP
[ 11948][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 11952][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[ 11952][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[ 12953][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 12959][I][main.cpp:1605] connectwifi(): [main] Try WiFi siatka
[ 14028][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: siatka, BSSID: 08:31:a4:8b:55:af, Channel: 5, Auth: WPA2_PSK
[ 14029][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 14082][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:10.0.0.167
[ 14083][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 14086][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 10.0.0.167, MASK: 255.255.255.0, GW: 10.0.0.2
[ 14160][I][main.cpp:1638] connectwifi(): [main] SSID = siatka
[ 14160][I][main.cpp:1642] connectwifi(): [main] IP = 10.0.0.167
[ 14160][I][main.cpp:2727] setup(): [main] Start web server
[ 14166][I][main.cpp:2737] setup(): [main] Network found. Starting clients
[ 14175][I][main.cpp:2771] setup(): [main] MDNS responder started
[ 15180][I][main.cpp:2816] setup(): [main] Rotary encoder is disabled (-1/-1/-1)
[ 15181][I][main.cpp:4372] gettime(): [main] Sync TOD
[ 15181][I][main.cpp:4385] gettime(): [main] TOD synced
[ 15288][E][main.cpp:749] nvsgetstr(): [main] nvs_get_str failed 1102 for key preset, keylen is 6, len is 150!
[ 15288][E][main.cpp:750] nvsgetstr(): [main] Contents:
[ 15292][I][main.cpp:966] nextPreset(): [main] nextPreset is 0
[ 15297][I][main.cpp:3403] radiofuncs(): [main] Radiofuncs cmd is 1
[ 15304][I][main.cpp:1400] connecttohost(): [main] Connect to host
[ 15309][E][AsyncTCP.cpp:729] connect(): error: -16
[ 15314][E][main.cpp:1472] connecttohost(): [main] Request  failed!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400edf5b  PS      : 0x00060d30  A0      : 0x800d88e8  A1      : 0x3ffcee30  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0xc4490906  A5      : 0x3ffcee10
A6      : 0x3ffcee10  A7      : 0x00000080  A8      : 0x800d645c  A9      : 0x3ffc60dc  
A10     : 0x3ffcebc8  A11     : 0x00000004  A12     : 0x3f400124  A13     : 0x000005c0
A14     : 0x3f401e86  A15     : 0x3f401f25  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000009  LBEG    : 0x4008bb95  LEND    : 0x4008bba5  LCOUNT  : 0xffffffff

Backtrace: 0x400edf58:0x3ffcee30 0x400d88e5:0x3ffcee50 0x400da1c8:0x3ffcee70 0x400f2269:0x3ffcee90

ELF file SHA256: 960efa8f48f0d366

E (15553) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

Edzelf, 2023-12-22:
Copy the files in the "data" directory to SPIFFS.
See [here](https://randomnerdtutorials.com/install-esp32-filesystem-uploader-arduino-ide/) for tutorial.

JureqB, 2023-12-22:
I did everything according to this tutorial: [](https://randomnerdtutorials.com/esp32-vs-code-platformio-spiffs/)
Now I have connected another new ESP32 Wroom 32D board, V1053 decoder and ILI19341 display:

Processing esp32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.4.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, 
tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 56 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Wire @ 2.0.0
|-- PubSubClient @ 2.8.0
|-- Adafruit BusIO @ 1.14.5
|-- Adafruit GFX Library @ 1.11.9
|-- Adafruit ILI9341 @ 1.6.0
|-- Adafruit ST7735 and ST7789 Library @ 1.10.3
|-- AsyncTCP @ 1.1.1
|-- ESP Async WebServer @ 1.2.3
|-- AC101 @ 0.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- codecs
|-- ILI9341
Building in release mode
Building FS image from 'data' directory to .pio\build\esp32\spiffs.bin
/about.html
/config.html
/defaultprefs.txt
/favicon.ico
/index.html
/mp3play.html
/nomp3play.html
/search.html
/style.css
================================ [SUCCESS] Took 5.12 seconds 

and next:

Processing esp32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.4.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 56 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Wire @ 2.0.0
|-- PubSubClient @ 2.8.0
|-- Adafruit BusIO @ 1.14.5
|-- Adafruit GFX Library @ 1.11.9
|-- Adafruit ILI9341 @ 1.6.0
|-- Adafruit ST7735 and ST7789 Library @ 1.10.3
|-- AsyncTCP @ 1.1.1
|-- ESP Async WebServer @ 1.2.3
|-- AC101 @ 0.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- codecs
|-- ILI9341
Building in release mode
Building FS image from 'data' directory to .pio\build\esp32\spiffs.bin
/about.html
/config.html
/defaultprefs.txt
/favicon.ico
/index.html
/mp3play.html
/nomp3play.html
/search.html
/style.css
Looking for upload port...
Using manually specified: COM6
Uploading .pio\build\esp32\spiffs.bin
esptool.py v4.5.1
Serial port COM6
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a8:42:e3:9e:af:34
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x002f0000 to 0x003fffff...
Compressed 1114112 bytes to 12288...
Writing at 0x002f0000... (100 %)
Wrote 1114112 bytes (12288 compressed) at 0x002f0000 in 5.5 seconds (effective 1617.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=============================================================================== [SUCCESS] Took 13.35 seconds

JureqB, 2023-12-22:
Unfortunately, nothing has changed: 

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (516) esp_core_dump_flash: No core dump f�ѥѥ�� found!
E (516) esp_core_dump_flash: No core dump partition found!
[     8][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3024][I][main.cpp:2582] setup(): [main] Version Thu, 14 Dec 2023 09:00:00 GMT.  Free memory 110580
[  3026][I][main.cpp:2583] setup(): [main] Display type is ILI9341
[  3123][I][main.cpp:2593] setup(): [main] SPIFFS is okay, space 1016801, used 33383
[  3126][I][main.cpp:2614] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3129][I][main.cpp:2614] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3138][I][main.cpp:2614] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1114112
[  3151][I][main.cpp:2486] fillkeylist(): [main] Read 0 keys from NVS
[  3155][I][main.cpp:1910] readIOprefs(): [main] pin_spi_sck set to 18
[  3160][I][main.cpp:1910] readIOprefs(): [main] pin_spi_miso set to 19
[  3166][I][main.cpp:1910] readIOprefs(): [main] pin_spi_mosi set to 23
[  3183][I][main.cpp:2655] setup(): [main] GPIO0 is HIGH
[  3193][I][main.cpp:2655] setup(): [main] GPIO2 is HIGH
[  3203][I][main.cpp:2655] setup(): [main] GPIO4 is HIGH
[  3213][I][main.cpp:2655] setup(): [main] GPIO5 is HIGH
[  3223][I][main.cpp:2655] setup(): [main] GPIO12 is HIGH
[  3233][I][main.cpp:2655] setup(): [main] GPIO13 is HIGH
[  3243][I][main.cpp:2655] setup(): [main] GPIO14 is HIGH
[  3253][I][main.cpp:2655] setup(): [main] GPIO15 is HIGH
[  3263][I][main.cpp:2655] setup(): [main] GPIO16 is HIGH
[  3273][I][main.cpp:2655] setup(): [main] GPIO17 is HIGH
[  3283][I][main.cpp:2655] setup(): [main] GPIO18 is HIGH
[  3293][I][main.cpp:2655] setup(): [main] GPIO19 is HIGH
[  3303][I][main.cpp:2655] setup(): [main] GPIO21 is HIGH
[  3313][I][main.cpp:2655] setup(): [main] GPIO22 is HIGH
[  3323][I][main.cpp:2655] setup(): [main] GPIO23 is HIGH
[  3333][I][main.cpp:2655] setup(): [main] GPIO25 is HIGH
[  3343][I][main.cpp:2655] setup(): [main] GPIO26 is HIGH
[  3353][I][main.cpp:2655] setup(): [main] GPIO27 is HIGH
[  3363][I][main.cpp:2655] setup(): [main] GPIO32 is HIGH
[  3373][I][main.cpp:2655] setup(): [main] GPIO33 is HIGH
[  3383][I][main.cpp:2655] setup(): [main] GPIO34 is LOW, probably no PULL-UP
[  3393][I][main.cpp:2655] setup(): [main] GPIO35 is LOW, probably no PULL-UP
[  3403][I][main.cpp:2655] setup(): [main] GPIO39 is LOW, probably no PULL-UP
[  3405][I][main.cpp:2672] setup(): [main] Start ILI9341 display
[  3406][E][main.cpp:2689] setup(): [main] Display not activated
[  3410][I][main.cpp:2278] mk_lsan(): [main] Create list with acceptable WiFi networks
[  3418][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: siatka
[  3425][I][main.cpp:2308] mk_lsan(): [main] Added siatka to list of networks
[  3948][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  4041][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[  4041][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[  4541][I][main.cpp:865] listNetworks(): [main] Scan Networks
[ 10755][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 9
[ 10756][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 10761][I][main.cpp:867] listNetworks(): [main] Scan completed
[ 10766][I][main.cpp:875] listNetworks(): [main] Number of available networks: 9
[ 10773][I][main.cpp:893] listNetworks(): [main]  1 - ArabicaGuest              Signal: -52 dBm, Encryption WPA_WPA2_PSK, Not accepttable
[ 10785][I][main.cpp:893] listNetworks(): [main]  2 - ARABICA2GHZ               Signal: -52 dBm, Encryption WPA2_PSK, Not accepttable
[ 10797][I][main.cpp:893] listNetworks(): [main]  3 - c96MQcTF                  Signal: -67 dBm, Encryption WPA2_PSK, Not accepttable
[ 10809][I][main.cpp:893] listNetworks(): [main]  4 - siatka                    Signal: -67 dBm, Encryption WPA2_PSK, Acceptable
[ 10820][I][main.cpp:893] listNetworks(): [main]  5 - sieciunia                 Signal: -68 dBm, Encryption WPA2_PSK, Not accepttable
[ 10832][I][main.cpp:893] listNetworks(): [main]  6 - G6_5308                   Signal: -71 dBm, Encryption WPA2_PSK, Not accepttable
[ 10843][I][main.cpp:893] listNetworks(): [main]  7 - CGA2121_nSeyQFz           Signal: -86 dBm, Encryption WPA2_PSK, Not accepttable
[ 10855][I][main.cpp:893] listNetworks(): [main]  8 - NETIASPOT-87EF10          Signal: -88 dBm, Encryption WPA_WPA2_PSK, Not accepttable
[ 10867][I][main.cpp:893] listNetworks(): [main]  9 - DIRECT-76-HP M477 LaserJet Signal: -90 dBm, Encryption WPA2_PSK, Not accepttable
[... 26 lines cut ...]
[ 15257][I][main.cpp:1400] connecttohost(): [main] Connect to host
[ 15262][E][AsyncTCP.cpp:729] connect(): error: -16
[ 15267][E][main.cpp:1472] connecttohost(): [main] Request  failed!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400edf5b  PS      : 0x00060d30  A0      : 0x800d88e8  A1      : 0x3ffcee30  
[... cut ...]
  #0  0x400edf58:0x3ffcee30 in VS1053::setVolume(unsigned char) at lib/codecs/src/VS1053.cpp:216
  #1  0x400d88e5:0x3ffcee50 in spfuncs() at src/main.cpp:3352
      (inlined by) spfuncs() at src/main.cpp:3327
  #2  0x400da1c8:0x3ffcee70 in loop() at src/main.cpp:3468
  #3  0x400f2269:0x3ffcee90 in loopTask(void*) at C:/Users/jtp/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 9ab732facf8772cc

E (15505) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...

JureqB, 2023-12-22:
I repeated this process once again, but now before pressing "Upload Filesystem Image" I selected the "Program Size" option. 
[image]
I uploaded everything again and... the website is now working, I'm starting to configure it.