Edzelf/Esp-radio #223: IDE version 1.8, esp8266 software 2.2.0, how to get LittleFS to work??

HoerMirAuf, 2024-10-29:
See the headline.

How to get LittleFS to work?
(LittleFS.h not found!!)

Edzelf, 2024-10-29:
I used:
```
name=LittleFS
version=0.1.0
author=Earle F. Philhower, III
maintainer=Earle F. Philhower, III <earlephilhower@yahoo.com>
sentence=Port of LittleFS to ESP8266 Arduino
paragraph=Replacement for SPIFFS to manage a filesystem in the onboard flash, supporting power fail safety and higher performance than SPIFFS at the cost of a lower maximum number of files.
category=Data Storage
url=https://github.com/esp8266/Arduino/libraries/LittleFS
architectures=esp8266
dot_a_linkage=true

```

HoerMirAuf, 2024-10-30:
Hi ... thank you for the URL.
But ist a 404 error.

The only LittleFS i can find is part of the Arduino Master tree:
https://github.com/esp8266/Arduino/tree/master/libraries
And this one will not compile with core 2.2.0

Maybe you can share your LittleFS package?

Edzelf, 2024-10-30:
I'm using version 3.1.1 of the ESP8266 toolchain:
```
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266WiFi 
Using library ESPAsyncTCP-master at version 1.2.0 in folder: D:\docs\Arduino\libraries\ESPAsyncTCP-master 
Using library ESPAsyncWebServer-master at version 1.2.0 in folder: D:\docs\Arduino\libraries\ESPAsyncWebServer-master 
Using library async-mqtt-client-master at version 0.8.1 in folder: D:\docs\Arduino\libraries\async-mqtt-client-master 
Using library SPI at version 1.0 in folder: C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\SPI 
Using library Adafruit_GFX_Library at version 1.11.5 in folder: D:\docs\Arduino\libraries\Adafruit_GFX_Library 
Using library Adafruit_BusIO at version 1.15.0 in folder: D:\docs\Arduino\libraries\Adafruit_BusIO 
Using library Wire at version 1.0 in folder: C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\Wire 
Using library TFT_ILI9163C-master at version 0.9 in folder: D:\docs\Arduino\libraries\TFT_ILI9163C-master 
Using library Ticker at version 1.0 in folder: C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\Ticker 
Using library ArduinoOTA at version 1.0 in folder: C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ArduinoOTA 
Using library TinyXML at version 1.0.2 in folder: D:\docs\Arduino\libraries\TinyXML 
Using library LittleFS at version 0.1.0 in folder: C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\LittleFS 
Using library Hash at version 1.0 in folder: C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\Hash 
Using library ESP8266mDNS at version 1.2 in folder: C:\Users\hp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266mDNS 

```
In the "Additional board manager URLs:
`http://arduino.esp8266.com/stable/package_esp8266com_index.json`

HoerMirAuf, 2024-10-30:
Thank you very much ... :)
3.1.1 toolchain, did it. 👍

What's about IwIP variant? i took v2 higher bandwich or it's better to take sometihng else?

Edzelf, 2024-10-30:
I don't know, just try.
Please consider [this](https://github.com/Edzelf/ESP32Radio-V2) for a better version of the radio.

HoerMirAuf, 2024-10-30:
Ok ... i did.
The only stable IwIP build seems to be: "v2 Lower Memory (no features)"
Thank you very much for your help and hints 👍

closed 2024-10-30 by HoerMirAuf