Edzelf/Esp-radio #226: output pin for external relay

ghostxder, 2025-05-08:
hello, I already have built the radio on breadboard, using a 2,8" display, the vs1053b, a 1838 IR module, rotary encoder and a chinese amplifier (12V).  I also changed the fontsize and clockposition (described in issue #160). It works perfekt at the moment. Now I want to establish an external line-in, which I can activate with the IR remote control (just change the amplifier input from vs1053b to the line in with a simple relay board). For this i need an output-pin defined which i can connect in the config to an ir-code (it will take two ir-codes like mute/unmute // relay on/relay off).
How can I manage this? Where/ in which file to define the output-pin? What else do I have to program?

Edzelf, 2025-05-09:
Add this to the function "const char* analyzeCmd ( const char* par, const char* val )", near line 4153:

```
  else if ( argument == "linein" )                    // Line input change?
  {
    pinMode ( xxxx, OUTPUT ) ;
    digitalWrite ( ivalue != 0 ) ;
  }
```
Where "xxxx" is the pin you want to use.
Now you can configure (in the preferences in the config page of the web interface) the 2 IR codes to execute "linein = 1" or "linein = 0", for example:

```
ir_807F = linein = 1
ir_9903 = linein = 0

```
Good luck!

ghostxder, 2025-05-09:
Thank you for the quick response! I first added the code you suggested, but I received an Error: "Zu wenig Argumente im Funktionsaufruf". Then I modified the digitalWrite see picture. I could upload the code, the radio is working and a LED connected to pin 13 shows a HI-Signal. When using the remote both buttons set pin 13 to LOW. Only way to get a HI-Signal again is to reset the radio. Unfortunately I can't figure out my mistake.

[image]

Edzelf, 2025-05-09:
Sorry, it should be:digitalWrite ( 13, ivalue != 0 ) ;Verzonden vanaf mijn Galaxy

Screenshot.2025-05-09.211336.png (view on web)

 .com>

[
{
"target": "https://github.com/Edzelf/Esp-radio/issues/226#issuecomment-2867694858",
"url": "https://github.com/Edzelf/Esp-radio/issues/226#issuecomment-2867694858",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

ghostxder, 2025-05-10:
Hello Ed, oh man... I thought much too complicated... didn't see the forest because of the trees in front of it...
Well, it is working, i can toggle the pin with the IR! 
After Reset/ Power up there is no defined state (low/ high) on the pin (I used others, it is the same there, also) and the test-LED light up, maybe half brightness (I connected the LED to ground including a 220 Ohm resistor). Defined pin states are after first use of IR.
Is this due to "SETUP - check programmable input pins" (~ line 2558 - 2570) ? Is there a chance to exclude this pin from being pulled up?
Would it be very complicated to display "Line in" when the relay-pin is high?

Markus

Edzelf, 2025-05-10:
Put the "pinMode" call into "setup()" at the end, followed by a call to digitaWrite to set the default state of the output pin.
Than you may removen the "pinMode" call from "analyzeCmd".

ghostxder, 2025-05-11:
Hello Ed, thank you so much, this solved the issue. The relay pin is low before music starts to play.

Markus

closed 2025-05-12 by Edzelf